RobotLocomotion/spartan #96: Build fails if eigen not system installed

manuelli, 2017-06-07:
If you don't have eigen system installed then the `spartan` build fails in the `ApriTags` part of the `director` superbuild. This is because it fails to find the eigen supplied by the drake build.

Potential solutions.

- add `libeigen3-dev` to the list of required dependencies.
- fix the apriltags build to find the drake supplied eigen.

peteflorence, 2017-06-08:
@manuelli is this for 16.04?

@patmarion why not just add to the 16.04 install prereqs script?

I have another suggestion for that 16.04 script, will pull request...

patmarion, 2017-06-08:
adding eigen to the prereqs is a workaround. i dont mind if you add it there.  this issue should remain open to track the bug in the april tags build system.

sammy-tri, 2017-06-13:
I think it's actually drake's Eigen doing something dumb here.  When it gets installed by the drake superbuild, the .pc file lands in `build/install/share/pkgconfig/eigen3.pc` (the only thing there).  If I manually copy it to `build/install/lib/pkgconfig/eigen3.pc` (where all of the other .pc files are) apriltags finds it and the build keeps going.

This workaround is rather infuriating, but that's what I've been doing...

manuelli, 2017-06-13:
Is it possible to get someone at TRI to volunteer some time to fix this issue? Or is it a won't fix due to upcoming Bazel switch?

sammy-tri, 2017-06-13:
I might get to it eventually, but I'm pretty slammed with corporate stuff at the moment (and my fear that the Eigen install code requires wandering into CMake world has kept me away so far).

peteflorence, 2017-10-04:
We are not having any issues with eigen in our Docker builds, meaning I think this issue has been resolved.  Please reopen if you think otherwise.

closed 2017-10-04 by peteflorence

manuelli, 2017-10-04:
We do system install `eigen` in the docker image I believe.

peteflorence, 2017-10-04:
Is there an issue with system installing eigen, or do you agree we can close?