RobotLocomotion/spartan #98: Why does spartan add drake directly AND indirectly via director?

stonier, 2017-06-08:
i.e. relevant parts of the CMakeLIsts.txt:

* [add_drake](https://github.com/RobotLocomotion/spartan/blob/master/CMakeLists.txt#L45)
* [director/USE_DRAKE=ON](https://github.com/RobotLocomotion/spartan/blob/master/CMakeLists.txt#L66)

manuelli, 2017-06-08:
@patmarion can answer this better but I believe it is because we are using the director superbuild. So we are telling director that drake is available, and hence to build the pieces that depend on drake. Director can be used in a standalone form (like drake-visualizer) where it doesn't depend on drake.

patmarion, 2017-06-08:
That's right.  The toplevel CMakeLists.txt in spartan uses CMake ExternalProject to build and install drake.  Then it uses ExternalProject to build director.  The `USE_DRAKE` option in director means that director should compile its optional features that include and link with drake's installed c++ libraries.

This design is intended to make it simple to switch to drake's bazel build when that is ready.  Director does not care how drake is built, as long as drake's installed components are available in the install directory.

stonier, 2017-06-13:
Thanks, my oversight :) Should have parsed the cmake.

closed 2017-06-13 by stonier