RobotLocomotion/spartan #101: "GMock" Error when building Spartan

spohorec, 2017-06-09:
Hi all, I am attempting to build Spartan on Ubuntu 14.04, and I am unable to fix the following error when running `make`:

`loading initial cache file /home/spohorec/spartan/build/tmp/RemoteTreeViewer-cache-Release.cmake
CMake Error at /usr/local/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  By not providing "FindGMock.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "GMock", but
  CMake did not find one.

  Could not find a package configuration file provided by "GMock" with any of
  the following names:
    GMockConfig.cmake
    gmock-config.cmake

  Add the installation prefix of "GMock" to CMAKE_PREFIX_PATH or set
  "GMock_DIR" to a directory containing one of the above files.  If "GMock"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /home/spohorec/spartan/build/install/lib/drake/cmake/drake-config.cmake:40 (find_dependency)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/spohorec/spartan/build/RemoteTreeViewer/CMakeFiles/CMakeOutput.log".
See also "/home/spohorec/spartan/build/RemoteTreeViewer/CMakeFiles/CMakeError.log".
make[2]: *** [src/RemoteTreeViewer-stamp/RemoteTreeViewer-configure] Error 1
make[1]: *** [CMakeFiles/RemoteTreeViewer.dir/all] Error 2
make: *** [all] Error 2
`
I have already installed all dependencies as instructed in the Spartan README, including google-mock. Any idea where this problem is coming from?

patmarion, 2017-06-09:
Hi Sarah.  Could you try enabling the cmake option `WITH_ISSUE_5456_WORKAROUND` in the spartan build?  You can do that like this:

```
cd spartan/build
cmake -DWITH_ISSUE_5456_WORKAROUND=ON ..
make
```

spohorec, 2017-06-09:
That seemed to work! Thanks!

closed 2017-06-09 by spohorec

manuelli, 2017-06-09:
Any errors related to find_package(drake) or RemoteTreeViewer require this fix as documented in the README. I am working on a fix to this at the moment.