RobotLocomotion/spartan #103: Build fails on Ubuntu 16.04 with VTK 5.10

5yler, 2017-06-12:
I installed VTK 5.10 from source (using [this tarball](https://github.com/Kitware/VTK/releases/tag/v5.10.0)) and tried building, and got the following error:
```
[ 84%] Built target ipopt_solver_test
/usr/bin/ld: /usr/local/lib/vtk-5.10/libvtkIO.a(vtkOBJReader.cxx.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/vtk-5.10/libvtkIO.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
systems/sensors/CMakeFiles/drakeSensors.dir/build.make:453: recipe for target 'systems/sensors/libdrakeSensors.so' failed
make[8]: *** [systems/sensors/libdrakeSensors.so] Error 1
CMakeFiles/Makefile2:13714: recipe for target 'systems/sensors/CMakeFiles/drakeSensors.dir/all' failed
make[7]: *** [systems/sensors/CMakeFiles/drakeSensors.dir/all] Error 2
make[7]: *** Waiting for unfinished jobs....
```

I then did `make clean` in `build` directory and tried to use `ccmake` to turn `USE_SYSTEM_VTK` off and set `VTK_DIR`:
[image]

That didn't seem to change anything, I got the exact same `ld` error as before, still in `/usr/local/lib/vtk-5.10/`. Please advise.

patmarion, 2017-06-12:
You should not need to manually compile vtk5 from source or install it to `/usr/local`.  I suspect you compiled vtk with static libraries and did not enable the required vtk options that will be needed in drake and director.

peteflorence, 2017-06-12:
Hi Syler, yes, for a 16.04 build, I would recommend just using the install_prereqs scripts, i.e. after cloning Spartan and running `git submodule update --init`, just running these commands: 

```
sudo ./drake/setup/ubuntu/16.04/install_prereqs.sh` 
sudo ./setup/ubuntu/16.04/install_prereqs.sh
mkdir build
cd build
cmake ../
make
```

If you try those, how far do you get?

5yler, 2017-06-12:
Removing the manually installed VTK 5.10 in `/usr/local/lib` and /usr/local/include`, then setting `USE_SYSTEM_VTK` to OFF and `VTK_DIR` to NOT FOUND seems to have worked!

(The `libvtk5-dev` package in the 16.04 `install_prereqs.sh` script conflicted with existing packages, including a significant portion of a working ROS install, so I didn't go that route)

closed 2017-06-12 by 5yler

peteflorence, 2017-06-12:
Glad that worked out.  Let us know if you have any other issues

patmarion, 2017-06-12:
You might still need to set the two environment variables defined here:

http://drake.mit.edu/faq.html?highlight=pythonpath#non-ros-users

As in something like thiat (but you might need to double check these paths are correct:

```
export LD_LIBRARY_PATH=$SPARTAN_BUILD_DIR/install/lib/vtk-5.10:$LD_LIBRARY_PATH
export PYTHONPATH=$SPARTAN_BUILD_DIR/director/src/vtk-build/Wrapping/Python:$PYTHONPATH
```
```