RobotLocomotion/spartan #104: Make Docker image for CORL software

peteflorence, 2017-06-12:
Everybody want this.  Targeting to due after June 28.

EricCousineau-TRI, 2017-06-15:
If it's a useful reference, [here is Brandon's PR](https://github.com/RobotLocomotion/drake/pull/6196) for adding Docker to Drake; I believe he is planning to test `nvidia-docker`, to run `drake-visualizer` from within the docker image.

peteflorence, 2017-06-15:
Thanks Eric!

manuelli, 2017-09-09:
Seems like this has happened already with the switch to LabelFusion.

patmarion, 2017-09-10:
Not exactly, the Docker image for LabelFusion only includes Director and ElasticFusion, not Spartan.  But the scripts for creating the LabelFusion Docker image could be easily ported to Spartan to create a Spartan docker image.

peteflorence, 2017-10-04:
Closing this issue since we now have:

- a dedicated [LabelFusion docker container](https://hub.docker.com/r/robotlocomotion/labelfusion/) in which LabelFusion works great
- [a full Spartan docker build](https://github.com/RobotLocomotion/spartan/blob/master/setup/docker/README.md).  Even though we haven't tested LabelFusion inside it.

closed 2017-10-04 by peteflorence