RobotLocomotion/spartan #112: LcmLogPlayer may not process all logged LCM events

EricCousineau-TRI, 2017-06-15:
I am not sure if I kludged my setup, but `sim_playback.py`, using `LcmLogPlayer`, does not seem to be finding the last `DRAKE_VIEWER_DRAW` timestamp with a Drake simulation using `drake::systems::DrakeVisualizer`.

If I record my simulation from the beginning (~10.5s of simulation time), then play it back, `LcmLogPlayer` will stop precessing `timestamps[-1] = 36600` (.366s) due to getting `None` as an event.
If I play the log back in `lcm-logplayer`, I can see the drawn simulation time go up to ~10.5s, and so if I change `director.lcmlogplayer` from:
```
    while True:
        # ...
        event = log.read_next_event()
        if not event:
            break
```
to:
```
    while lastEventTimestamp < 10.5e6:
        # ...
        event = log.read_next_event()
        # if not event:
        #    break
```
then it will get up to 10.5s in the recorded simulation.

The only slightly relevant change is that I briefly modified `drake::systems::DrakeVisualizer` on my `dev` branch to add a prefix to LCM channels, and made an extension of `director`'s `DrakeVisualizer` that will inject prefixes to the LCM channels (as a quick way to visualize both the simulated and estimated plant from `drake`).

* [Example with multiple visualizers](https://github.com/EricCousineau-TRI/drake/blob/feature/dart_reimpl-wip/drake/examples/kuka_iiwa_arm/dev/monolithic_pick_and_place/helper_app.py#L47)

I will see if I can trim the existing log and post that.

patmarion, 2017-06-15:
you're saying that when reading your log `log.read_next_event()` returns `None` before the end of the log, and there are more messages that can be read after the `None`?

EricCousineau-TRI, 2017-06-15:
Yup, at least that's what commenting out that line seems to imply.
I haven't done much further digging past that.

peteflorence, 2017-10-05:
@EricCousineau-TRI this looks like still an issue... what are you thinking we should do?

EricCousineau-TRI, 2017-10-25:
I have still used a workaround to this to get deterministic playback, but that was namely because I had an existing hacky solution that worked.

If you'd like, I can close the issue, but I'd like to ensure that the title mentions being "[UNSOLVED]".