RobotLocomotion/spartan #117: Improve support for private drivers

patmarion, 2017-07-07:
If you build spartan source code and set `WITH_DRIVERS=1` then the superbuild attempts to checkout and build proprietary driver source code which is hosted in private git repositories.

Normally I only build `WITH_DRIVERS=1` on maches in the robot lab.  Those machines have the necessary ssh keys setup to access the private drivers.

When external collaborators attempt to use spartan with their robot hardware there is no simple way for them to build the drivers using alternate repositories, they need to have access to the repos that are listed in the spartan superbuild.

peteflorence, 2017-10-04:
@patmarion is there still an easier way you had in mind to make this possible?

master branch now includes options for both WITH_IIWA_DRIVER_RLG and WITH_IIWA_DRIVER_TRI, which is maybe enough of example code to show how proprietary drivers could be involved?

manuelli, 2018-06-25:
We are moving away from public support for `spartan`. Only the RLG use case is supported for now.

closed 2018-06-25 by manuelli