RobotLocomotion/spartan #124: missing drake submodule snopt

YuehChuan, 2017-09-08:
[image]

YuehChuan, 2017-09-08:
temperately I remove the submodule seems it optional nonlinear optimization library
https://gist.github.com/kyleturner/1563153 
and turn off CMake configure
It would be good to add that in organization repo, thanks!

manuelli, 2017-09-09:
`snopt` is a private submodule not accessible to outside users. You can turn off the `WITH_SNOPT` option in the `drake` build to fall back to using `ipopt`. In general `spartan` is meant for internal use and we won't be providing support to external users for the time being.

closed 2017-09-09 by manuelli

YuehChuan, 2017-09-09:
It got compiled! Thanks for making this project available!