RobotLocomotion/spartan #126: can't find GMock during find_package(drake)

manuelli, 2017-09-21:
When building the kuka driver on a clean 16.04 install I ran into the error

> loading initial cache file /home/locomotion/spartan-2/build/tmp/drake-iiwa-driver-cache-Release.cmake
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  By not providing "FindGMock.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "GMock", but
  CMake did not find one.

  Could not find a package configuration file provided by "GMock" with any of
  the following names:

    GMockConfig.cmake
    gmock-config.cmake

  Add the installation prefix of "GMock" to CMAKE_PREFIX_PATH or set
  "GMock_DIR" to a directory containing one of the above files.  If "GMock"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /home/locomotion/spartan-2/build/install/lib/drake/cmake/drake-config.cmake:40 (find_dependency)
  kuka-driver/CMakeLists.txt:2 (find_package)

It seems that this is erroring out on `find_package(drake)` [here]{https://github.com/RobotLocomotion/drake-iiwa-driver/blob/master/kuka-driver/CMakeLists.txt#L2}. @sammy-tri have you encountered this issue before? By the way I am building the `drake` inside `spartan` using CMake, not Bazel so this could be related to all the `find_package(drake)` issues we have had in the past.

gizatt, 2017-09-21:
I'm not sure if it's in the same context (i.e. inside of an external project) but when I've encountered this issue, I've resolved it by explicitely adding `drake/cmake/modules` to the CMake module path for the external projects that require it. Cludgy, but it cleared it for me. See https://github.com/RobotLocomotion/spartan/commit/ad469bec269487ab1c6927334d4d6768c456da42

manuelli, 2017-09-28:
fixed by #116

closed 2017-09-28 by manuelli