RobotLocomotion/spartan #132: setting WITH_SNOPT ON breaks build inside docker container

manuelli, 2017-10-03:
This results in the error

```
ERROR: /root/spartan/drake/drake/solvers/BUILD:477:1: no such package '@snopt//': Traceback (most recent call last):
    File "/root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external/bazel_tools/tools/build_defs/repo/git.bzl", line 67
        _clone_or_update(ctx)
    File "/root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external/bazel_tools/tools/build_defs/repo/git.bzl", line 42, in _clone_or_update
        fail("error cloning %s:\n%s" % (ctx.n...))
error cloning snopt:
+ cd /root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external
+ rm -rf /root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external/snopt
+ git clone git@github.com:RobotLocomotion/snopt.git /root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external/snopt
Cloning into '/root/.cache/bazel/_bazel_root/2561d68fd8fc2694a44484d8a10a4070/external/snopt'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
 and referenced by '//drake/solvers:snopt_solver'.
ERROR: Analysis of target '//:install' failed; build aborted.
INFO: Elapsed time: 0.459s
```
Looks like the `ssh` file doesn't exist. Also `git` is not installed. Probably we need to copy `~/.ssh` from outside the container to inside.

patmarion, 2017-10-03:
We could change to https, but then you will have to interactively type your github password at build time.

To use the ssh url we'll need to install ssh client inside the docker image: `apt-get update; apt-get install openssh-client`.

Next, to use ssh keys, I wouldn't copy `~/.ssh` into the image when building, although that is an option.  What I do in my docker dev environment is mount my whole home directory, but for this the user could just mount `~/.ssh` using the `-v` arg to docker run, as in, `-v ~/.ssh:/root/.ssh`.  But this will fail because of a permissions issue, since files inside `.ssh` are read only by your user, not by the root user in docker.  Ideally we'd stop using the root user in docker and setup a script so that the user name and id inside docker is the same as outside docker, then things run pretty smoothly.

patmarion, 2017-10-03:
There might also be a way to share ssh keys withouth mounting files, using ssh-agent, but I haven't looked into how to do that with docker.

manuelli, 2017-10-03:
I'm happy mounting the `~/.ssh` and having it so that the user insider the docker is the same as outside and is not root. I know how to do the mounting. I'll poke around for how to do the user thing tomorrow.

peteflorence, 2017-10-03:
@patmarion super helpful, thank you!

The Dockerize Spartan PR (https://github.com/RobotLocomotion/spartan/pull/130/commits/fc9cb22a69ee1aebe44ed65f2ebe1eba5bed1a26) fixes this:
- adds `ssh` to build dependencies of docker: https://github.com/RobotLocomotion/spartan/pull/130/commits/130670e444c5e1a5b0abd3afa9fb2ebb11b704b5
- mounts ssh keys: https://github.com/RobotLocomotion/spartan/pull/130/commits/fc9cb22a69ee1aebe44ed65f2ebe1eba5bed1a26

This has been tested (completing a build with SNOPT now).

Turned out to not need user permissions, although does sound better and would like to get that set up soon.

closed 2017-10-03 by peteflorence