RobotLocomotion/spartan #133: cameraview.py looking for ddBotImageQueue

peteflorence, 2017-10-03:
This is a tool I loved to use for LabelFusion.

`directorPython -m director.imageviewapp --bot-config $SPARTAN_SOURCE_DIR/config/bot_frames.cfg --channel OPENNI_FRAME --rgbd --pointcloud`

But `cameraview.py` is throwing an error (see below).

I think the fix here is to merge some LabelFusion edits (on `pf/corl-master` branch of Director) into Director master.

I'll look into fixing now.

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/root/spartan/build/install/lib/python2.7/site-packages/director/imageviewapp.py", line 124, in <module>
    main()
  File "/root/spartan/build/install/lib/python2.7/site-packages/director/imageviewapp.py", line 120, in main
    ImageViewApp().start()
  File "/root/spartan/build/install/lib/python2.7/site-packages/director/imageviewapp.py", line 18, in __init__
    self.setup()
  File "/root/spartan/build/install/lib/python2.7/site-packages/director/imageviewapp.py", line 41, in setup
    imageManager = cameraview.ImageManager()
  File "/root/spartan/build/install/lib/python2.7/site-packages/director/cameraview.py", line 116, in __init__
    self.queue = PythonQt.dd.ddBotImageQueue(lcmUtils.getGlobalLCMThread())
AttributeError: 'module' object has no attribute 'ddBotImageQueue'
```

gizatt, 2017-10-03:
I think most point cloud visualization in Spartan via Director will require those changes merged!

peteflorence, 2017-10-03:
:+1: working on it

peteflorence, 2017-10-03:
this is on hold until we first get `run_tests_director` passing on `master` of `spartan` (currently failing)

peteflorence, 2017-10-05:
This will be fixed by https://github.com/RobotLocomotion/director/pull/565

PR #146 (2017-10-06): update director submodule, pass USE_PERCEPTION flag to director build
- update the director submodule
- pass the USE_PERCEPTION flag to the director build.

@gizatt 

fixes #133

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/spartan/146)
closed 2017-10-06 by manuelli