RobotLocomotion/spartan #139: Launch drake executables with `bazel run` syntax

manuelli, 2017-10-05:
Currently we have the short term solution of copying the executables from the `drake` `.bazel` folder as in #138

rdeits, 2017-10-05:
FYI: `bazel run` for one command will block all other `bazel run` and `bazel build` commands on your system, so this might not be a good idea: https://github.com/RobotLocomotion/drake/issues/6855

fbudin69500, 2017-10-05:
Why not running the executables that are installed? Am I missing a piece of information?

manuelli, 2017-10-05:
I don't think that all the executables we want are currently installed. For example `kuka_simulation` or `iiwa_wsg_simulation`

gizatt, 2017-10-05:
That's a thing that we should fix as we figure out what we need!

My sim development branch will have us compiling our own versions of those files, btw, as I'm making some modifications to what they broadcast / how they set up their environment.

manuelli, 2018-07-12:
As of #310 we will no longer be consuming any `drake` executables.

closed 2018-10-02 by manuelli