RobotLocomotion/spartan #148: director segfaults after a while when xtion connected

manuelli, 2017-10-06:
We get this error after `kuka_iiwa_app` is running for a while. I think this is an issue decoding the images coming from the xtion. This is happening consistently.

```
unsupported marker type 0x73
/home/manuelli/spartan/scripts/bin/kuka_iiwa_app: line 4: 20220 Segmentation fault      (core dumped) 
```

gizatt, 2017-10-06:
I've dug a bit on this one before -- it happens during jpeg decompression. As a workaround till we get to the bottom of it, you can turn off jpeg compression in the openni driver (I think it's `-j 100`), and the issue will go away (though you'll need to make sure the UDP buffer is big enough that message still get through).

patmarion, 2017-10-06:
I think there was a crash in elasticfusion, using the jpeg decompress code from elasticfusion, but executed in data from the lcm openni driver.  Maybe the issue is in the compression method, and not the decompress?

gizatt, 2017-10-06:
Definitely could be compression rather than decompression (or compression assuming a standard that the decompression doesn't agree with?). Just to put it out there, a slightly-deeper stopgap could be to allow the unsupported marker error to happen, but catch it and ignore that whole rgb-d frame rather than crashing.

manuelli, 2018-06-25:
Has become a non-issue due to using ROS driver

closed 2018-06-25 by manuelli