RobotLocomotion/spartan #150: director freezes when moving pointcloud

manuelli, 2017-10-06:
The symptom is that if you have a pointcloud visualized, add a child frame to it with `vis.addChildFrame()` and then try to move that child frame, director freezes, in fact the whole computer freezes. The keyboard stops working and I end up having to do a reboot.

I am guessing this is related to some API change in vtk8/Qt5. @gizatt @patmarion

manuelli, 2017-10-06:
I am able to create a sphere, add a child frame to it and move it around no problems. So the issue is when adding a child frome to an DisparityPointCloudItem

patmarion, 2017-10-06:
Sounds like it could be an issue with vtk together with the ubuntu16 system version of qt5.

manuelli, 2017-10-06:
Any idea of where to start debugging it?

patmarion, 2017-10-06:
I would start by collecting a backtrace and use print statements to locate the crash.

I would also reconfigure the Director superbuild options to use vtk7 and qt4 and verify that the crash does not exist in that configuration.