RobotLocomotion/spartan #152: Merge ROS interop ability from Draper Momap Team

gizatt, 2017-10-10:
Their codebase has tons of modules, many of which we aren't going to use right away (e.g. segicp) -- the principle thing I'd like to take is their LCM <-> ROS translation layer, that lays all the groundwork to make the IIWA work cleanly in ROS, and their control utilities offering a gotoconf service.

But their solution is pretty specifically tailored to their setup, and ought to be generalized. I'm going to collect a list of the utilities that I'm bringing in, and the changes I think are necessary to generalize them, [here](https://docs.google.com/document/d/1sdgWZIR_REkmZSqnw51XlQyaHgvZWMnN1jIHLsvnFCU/edit?usp=sharing).

dmsj, 2017-10-13:
@gizatt I'm tagging @jaymwong and @mrhebert here. We'd like to help make this be less painful and have the final implementation still work with our system.

dmsj, 2017-10-13:
@gizatt which kuka driver are you using? have you built a version of kuka_driver.cc in bazel drake? This links to `lcmtypes/drake/lcmt_iiwa_command.lcm` etc... and we haven't built it into a standalone executable. we haven't migrated to bazel drake yet.

gizatt, 2017-10-13:
Great! I assume you found the answer to that deleted post? `kuka-driver` comes in from `drake-iiwa-driver` as an external project (a Spartan-specific fork that isn't bazel yet, though we ought to get back to master soon), as found by the top-level `CMakeLists.txt`.

My branch `gizatt-with-ros` that is PR'd has some very pared-down versions of the core `robot-control` and `robot-server` utilities -- they serve our desire to sprint to some simple ROS functionality. I'm definitely behind you on eventually replacing those with generalized versions that support both of our systems, though.

I'm editing the above-mentioned google doc with some higher-level constraints I'm thinking of that need to be satisfied for the drivers to work for both systems.

dmsj, 2017-10-13:
yeah we fixed it - it works with bazel-drake now. turns out it wasn't a big deal. We'll take a look at your PR branch.

jaymwong, 2017-10-13:
@gizatt; the stripped down `robot_server` and `robot_control` ROS packages look functional to me at least in `spartan/gizatt-with-ros`. Let me know if you need any of my help. 

To answer some of your question, `generalized_robot` was intended to be our main interfaces into `drake` functionality when connecting to `hpn` and the rest of our ROS stack. It handles the construction of the `hpn` compatible robot representation and provide hooks into the robot's inverse kinematics through `drake`. A ton of stuff is built on top of it which is likely why it's so confusing and eclectic. 

Decoupling the `robot_control/GoToConf` service from `generalized_robot` should be pretty painless; let me know if you're having problems, if you are, I could do it in a branch for you.  

We started cleaning up our robot configurations (e.g. robot settings with gripper, with different end effectors, or with pan-tilt, etc) a while ago. See our `robot_descriptions` repo for the associate `xacro`-compatible urdfs. We've been meaning to perhaps have `robot_server` read from these xacros to determine what components are actually in the system when populating the `joint_state`...though, I haven't gotten around to it yet. 

In response to,
```Pub_joint_state.py depends on generalized robot in a way I don’t currently understand (for requesting configurations), because I don’t know what generalized_robot is. Has been commented out for now
```
You can probably remove anywhere that says `defaultConf` as this was a trick to initial the real system at whatever state `hpn` was in. A terrible feature (or bug).

manuelli, 2017-10-13:
We just got our robot moving by sending messages from ROS --> LCM --> kuka_plan_runner --> kuka_driver --> Robot. I chaned `trajectory_server.py` to make it much simpler. It seems there was some strange rescaling happening in there before. Now it's just a straight translation of the `JointTrajectory` message to a `robot_plan_t` message.

peteflorence, 2017-11-13:
@manuelli @gizatt do you agree we can close this, now that the robots move with the LCM<-->ROS translation?  Or are there other specific components we still want to bring in?

manuelli, 2017-11-13:
I'm happy closing this.

closed 2018-03-15 by manuelli