RobotLocomotion/spartan #161: kuka driver seg faulting

peteflorence, 2017-10-12:
What we have so far:

```
peteflo@e8230ed132e3:~/spartan/build$ gdb --args kuka_driver
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kuka_driver...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/peteflo/spartan/build/install/bin/kuka_driver 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff69cd700 (LWP 8427)]
Listening for robot 0 port 30200
LBRiiwaClient state changed from 0 to 1
onStateChange ( 1507817757199000): old 0 new 1
onStateChange ( 1507817757199000): quality 0 safety 0 oper 2 drive 0 control 0 command 0 overlay 0

Thread 1 "kuka_driver" received signal SIGSEGV, Segmentation fault.
0x000000000040a3c8 in kuka_driver::do_main() ()
(gdb) bt
#0  0x000000000040a3c8 in kuka_driver::do_main() ()
#1  0x0000000000406043 in main ()
(gdb) 

```

peteflorence, 2017-10-12:
With debug flags

```
peteflo@e8230ed132e3:~/spartan/build$ gdb --args kuka_driver
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kuka_driver...done.
(gdb) run
Starting program: /home/peteflo/spartan/build/install/bin/kuka_driver 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff69cd700 (LWP 11248)]
Listening for robot 0 port 30200
LBRiiwaClient state changed from 0 to 1
onStateChange ( 1507817990714000): old 0 new 1
onStateChange ( 1507817990714000): quality 0 safety 0 oper 2 drive 0 control 0 command 0 overlay 0

Thread 1 "kuka_driver" received signal SIGSEGV, Segmentation fault.
0x0000000000411eaa in __int64_t_encode_array (_buf=0x6952d0, offset=76, maxlen=336, p=0x0, elements=7)
    at /home/peteflo/spartan/build/install/include/lcm/lcm_coretypes.h:277
277	        const uint64_t v = unsigned_p[element];
(gdb) bt
#0  0x0000000000411eaa in __int64_t_encode_array (_buf=0x6952d0, offset=76, maxlen=336, p=0x0, elements=7)
    at /home/peteflo/spartan/build/install/include/lcm/lcm_coretypes.h:277
#1  0x0000000000412162 in __double_encode_array (_buf=0x6952d0, offset=76, maxlen=336, p=0x0, elements=7)
    at /home/peteflo/spartan/build/install/include/lcm/lcm_coretypes.h:365
#2  0x00000000004132a1 in drake::lcmt_iiwa_status::_encodeNoHash (this=0x7fffffffdfc0, buf=0x6952d0, offset=8, maxlen=404)
    at /home/peteflo/spartan/build/install/include/drake/lcmtypes/drake/lcmt_iiwa_status.hpp:146
#3  0x00000000004130c1 in drake::lcmt_iiwa_status::encode (this=0x7fffffffdfc0, buf=0x6952d0, offset=0, maxlen=412)
    at /home/peteflo/spartan/build/install/include/drake/lcmtypes/drake/lcmt_iiwa_status.hpp:93
#4  0x0000000000414c86 in lcm::LCM::publish<drake::lcmt_iiwa_status> (this=0x7fffffffdf98, channel="IIWA_STATUS", msg=0x7fffffffdfc0)
    at /home/peteflo/spartan/build/install/include/lcm/lcm-cpp-impl.hpp:157
#5  0x0000000000413db2 in kuka_driver::KukaLCMClient::PublishStateUpdate (this=0x7fffffffdf90)
    at /home/peteflo/spartan/build/externals/drake-iiwa-driver/kuka-driver/kuka_driver.cc:134
#6  0x00000000004125b9 in kuka_driver::do_main ()
    at /home/peteflo/spartan/build/externals/drake-iiwa-driver/kuka-driver/kuka_driver.cc:306
#7  0x0000000000412723 in main (argc=1, argv=0x7fffffffe1e8)
    at /home/peteflo/spartan/build/externals/drake-iiwa-driver/kuka-driver/kuka_driver.cc:320
```

peteflorence, 2017-10-12:
Fixed with help from @gizatt, in drake-iiwa-driver: https://github.com/RobotLocomotion/drake-iiwa-driver/commit/35909735606d96d0bdf3fd9895b534d50360a77e

Added a repoint of the drake-iiwa-driver sha in https://github.com/RobotLocomotion/spartan/pull/160/commits/27d53e18dd248bf6f0f7a59c608475c55de82ae0

Needed to match new lcm message type that includes joint_velocity_estimated @sammy-tri

closed 2017-10-12 by peteflorence

sammy-tri, 2017-10-13:
FWIW, this was fixed on master in 3939e32, though I probably should have made it more obvious by having a separate commit for it.