RobotLocomotion/spartan #171: jenkins errors out if another container with the same name on the machine

manuelli, 2017-10-13:
```
docker: Error response from daemon: Conflict. The container name "/spartan" is already in use by container "956d5e8d16f24525f78ac5d85f6142079a9ad7c72f475b73264f2a175bf1f771". You have to remove (or rename) that container to be able to reuse that name.
```
This error was caused because another container named `spartan` already exists. I think we should not pass the `--name` flag to `docker run` when doing Jenkins. This will always create a new name.

gizatt, 2017-10-14:
Good catch, I'll fix that up when I'm at a computer again in an hour or
two. Never had two long running (i.e. working) builds running before!

gizatt, 2017-10-14:
I've switched Jenkins to use unique image names for each of the two jobs, but that isn't enough, as each docker image wants to bind the same ports. There's little point to doing two builds on one machine anyway, given that that machine only has 8 cores, and they are well used by the build, so I've just disabled trying to build more than one job at a time. So this shouldn't happen again.

closed 2017-10-14 by gizatt