RobotLocomotion/spartan #173: bash/terminal completion not working in docker

manuelli, 2017-10-16:
If you do `rostopic echo /cam . . . ` and try to tab complete it doesn't work.

gizatt, 2017-10-17:
I've tried adding `apt-get install --reinstall bash-completion`, tried installing `bash-completion-el`, tried sourcing `/usr/share/bash-completion/bash_completion` and `/etc/bash_completion` from `entrypoint.sh`, and I can't figure out why this doesn't work.

There appear to be two places that the `complete` builtin gets its info:
- `/etc/bash_completion` with config in `/etc/bash_completion.d`, which has only a few files by default
- `/usr/share/bash-completion/bash_completion`, with config next to it, which has many files support apt-get and other things installed.

Maybe they're not getting loaded somehow? Or the wrong one is being loaded (the one without all of the support for installed applications? Because bash completion work for very simple commands, just not for ROS or apt-get or other things that need specific support).

peteflorence, 2017-10-17:
I have a fix for `apt-get install ....`

I have not looked into ros bash completions

closed 2017-10-17 by gizatt