RobotLocomotion/spartan #175: Sourcing setup.sh from catkin_build blows away ROS_PACKAGE_PATH changes

peteflorence, 2017-10-17:
See current solution here: https://github.com/RobotLocomotion/spartan/blob/master/config/setup_environment.sh.in#L25

siyuanfeng-tri, 2017-10-19:
also if you didn't ccmake with ros, 
https://github.com/RobotLocomotion/spartan/blob/master/config/setup_environment.sh.in#L28
cause unhappiness.

gizatt, 2017-10-19:
Good point, sorry about that -- we should definitely add some check-if-file-exists logic to avoid that.

gizatt, 2017-10-19:
Or, check-if-we-built-with-ROS, rather. Don't want ROS brought in unless the user asked for it to be done

peteflorence, 2017-10-19:
@siyuanfeng-tri yes there is some general clean-up to do regarding setting options (ROS: {yes, no}, ElasticFusion: {yes, no})

@siyuanfeng-tri is anything broken for you now?

siyuanfeng-tri, 2017-10-19:
I can't source that file because L28 doesn't exist.

peteflorence, 2017-10-20:
I am renaming this issue and broadening scope to incorporate the variety of issues we have

- [ ] Sourcing setup.sh from catkin_build blows away ROS_PACKAGE_PATH changes
- [ ] Sourcing labelfusion is required by current `kuka_iiwa_app` on master
- [ ] Installing ros is also required by current `kuka_iiwa_app` on master
- [ ] There is no option to not build elasticfusion

This is a result of quickly merging in new stuff, which I'm glad we did, but we just need to do some cleanup.

manuelli, 2018-03-15:
Seems like this hasn't been an issue in a while. Closing

closed 2018-03-15 by manuelli