RobotLocomotion/spartan #176: Can't start director with spartan master.

siyuanfeng-tri, 2017-10-19:
I also run into this problem when starting director. 
my director sha 02c2ef65f8d1d9f3de1d56d129351cd43846d70b
drake sha 2d606c778396ddd22d61e08516e7bc2b98ad8b56
Any ideas??? 

[2017-10-19 15:58:00.907] [console] [warning] Couldn't find package 'iiwa_description' in the supplied packagepath.
[2017-10-19 15:58:00.907] [console] [warning] Mesh 'package://iiwa_description/meshes/visual/link_0.obj' could not be resolved and will be ignored by Drake. If you don't want it to be ignored, please include it in the package map.
Traceback (most recent call last):
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/consoleapp.py", line 51, in onStartup
    func()
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/mainwindowapp.py", line 326, in loadScripts
    execfile(filename, fields.globalsDict)
  File "iiwaManipApp.py", line 254, in <module>
    robotSystem = makeRobotSystem(view)
  File "iiwaManipApp.py", line 106, in makeRobotSystem
    robotSystem = factory.construct(view=view, options=options)
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/componentgraph.py", line 148, in construct
    self.initComponent(name, defaultFields)
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/componentgraph.py", line 164, in initComponent
    newFields = initFunction(inputFields)
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/robotsystem.py", line 62, in initRobotState
    visible=True)
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 205, in loadRobotModel
    model = loadRobotModelFromFile(urdfFile)
  File "/home/user/code/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 246, in loadRobotModelFromFile
    if not model.loadFromFile(filename, floatingJointType):
RuntimeError: std::runtime_error: drake/multibody/parsers/urdf_parser.cc: ParseGeometry: ERROR: Mesh file name could not be resolved from the provided uri "package://iiwa_description/meshes/visual/link_0.obj".

Interrupt

peteflorence, 2017-10-19:
Hi Siyuan,

Is this what you get when you call kuka_iiwa_app? Can debug in morning

siyuanfeng-tri, 2017-10-19:
yep, thanks pete!

peteflorence, 2017-10-20:
@siyuanfeng-tri will close and comment on fix.  thanks!

siyuanfeng-tri, 2017-10-20:
I commented out the last line in build/setup_enviroment.sh because I didn't want any ros stuff yet, which cause director to not knowing where to look for the model files. After putting this back, director can load robot models. But it does have other complaints about labelfusion and subsequent ros dependencies, which doesn't affect my simplistic use case of director. 

export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$SPARTAN_SOURCE_DIR/drake/drake/manipulation/models:$SPARTAN_SOURCE_DIR/models

closed 2017-10-20 by siyuanfeng-tri

peteflorence, 2017-10-20:
Thanks @siyuanfeng-tri.  See https://github.com/RobotLocomotion/spartan/issues/175 for tracking the general cleanup