RobotLocomotion/spartan #178: problem with schunk gripper visualization in RVIZ

manuelli, 2017-10-31:
RVIZ throws an error about not being able to find the transform from the schunk gripper fingers to base.

```
wsg_50_finger_left_link
No transform from [wsg_50_finger_left_link] to [base]
```

gizatt, 2017-11-30:
Problem is that no one is publishing that information. It looks like the schunk state is currently publishing a custom state message (defined in wsg50_msgs) -- it'll have to also publish a JointState message (for Draper, `robot_server/pub_joint_state.py` did this by subscribing to the LCM state of both robot and gripper). Is there a more natural place in our system to do that than in the schunk driver itself?

manuelli, 2017-12-01:
Does the /joint_state message need to icnlude information for all the joints, or can we publish several of them that get put together?

gizatt, 2017-12-01:
I'm not sure -- I'm converting sim to generate ROS messages equivalent to the gripper, so I'll be able to try that out once that sim works. I hope (and would suspect) that we can find a way to make it work from multiple channels. (Or maybe we can publish on the same channel, but about different joints? Since the `joint_state` message includes both the joint names and joint angles?)

manuelli, 2018-03-15:
Seems like the schunk gripper is visualizing fine now.

closed 2018-03-15 by manuelli