RobotLocomotion/spartan #180: Drake and ROS integration for running the iiwas

manuelli, 2017-11-02:
@stonier I think you are leading the ROS + drake integration charge over there. We have developed a pretty decent ROS + `spartan` setup for running our iiwa. Maybe it's worth syncing up in person at some point to discuss future roadmap. I am creating this issue as a sandbox for discussing ideas.

@peteflorence @gizatt @siyuanfeng-tri @sammy-tri

stonier, 2017-11-07:
> From TRI's side, I am currently building a parallel repo along the lines
> of https://github.com/RobotLocomotion/drake-shambhala which links to
> drake and will include ROS message pub / sub system. The initial plan is to
> have a Kinova sim in this repo which natively talks ros and can be
> visualized in rviz. I will annouce this repo later this week when it should
> be ready.
> I have closely followed the tools landing in spartan and have discussed
> some of it with @gizatt <https://github.com/gizatt>.
> I would like to join this roadmap sync'up too

Sounds good. Give me a week or two and I can probably give a report out on
where the remaining issues (mostly conflict) are. Right now there are one
or two unknowns to de-risk.

Daniel

robinzhoucmu, 2017-11-08:
So far, my biggest issue is I couldn't use catkin_make while linking to drake::drake. 
It's the same -loctomap issue. 

I have been able to use a separate repo and treat both drake and ros as external library within my own CMakelists.

manuelli, 2018-06-25:
We have ROS integration now.

closed 2018-06-25 by manuelli