RobotLocomotion/spartan #189: Add ROS driver for schunk gripper

manuelli, 2017-11-09:
@sammy-tri any thoughts?

sammy-tri, 2017-11-09:
I have no use for such a thing.

sammy-tri, 2017-11-09:
That may have been unnecessarily harsh...   The LCM based one is working fine for me and I doubt I'll want to control the gripper using ROS any time in the near future.

manuelli, 2017-11-09:
Ok, we'll take this over. I think our codebases are rapidly diverging so just wanted to give you the heads up.

manuelli, 2017-11-16:
Looks like a driver already exists here https://github.com/nalt/wsg50-ros-pkg

manuelli, 2018-07-26:
After talking with a visitor we should use the [nalt](https://github.com/nalt/wsg50-ros-pkg) driver. They have been using it successfully. @gizatt do you want to take the lead on this?

manuelli, 2018-07-26:
That's fine, I'm in no rush. We have been using it like this so far, but better to eliminate the red error light from our gripper.

gizatt, 2018-08-21:
I've perused a few of the options with the nalt driver, and I think the Xamla fork is the most modern / usefully-featured. It's a bit of a mess (it exposes two different actionlib servers with different interfaces and has zero documentatino), but it allows functional enough position, speed, and force control with real-time feedback. I'll start switching things over to that driver.

manuelli, 2018-11-05:
This is in master

closed 2018-11-05 by manuelli