RobotLocomotion/spartan #204: docker_run only compatible with python 2

gizatt, 2017-11-30:
```
  File "./setup/docker/docker_run.py", line 23
    print "running docker container derived from image %s" %args.image
                                                         ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("running docker container derived from image %s" %args.image)?
```

gizatt, 2017-11-30:
To be clear: not proposing a massive compat overhaul for python 3, just noting one issue that makes invoking `docker_run` a little more annoying :)

manuelli, 2017-11-30:
AFAIK the `print` statment is the big difference between Python2 and Python3. If there is some sort of print statment that works for both of them I am happy to switch to that. Also wrapping in try/catch is fine.

rdeits, 2017-11-30:
At the top of the file, add:

```python
from __future__ import print_function
```

then change all `print "foo"` to `print("foo")` (i.e. python3 style) and the code will work in both Python2 and Python3

manuelli, 2017-11-30:
Thanks for the tip Robin!

PR #205 (2017-11-30): Switch prints to backward-compatible syntax
Resolves #204

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/spartan/205)
closed 2017-11-30 by manuelli