RobotLocomotion/spartan #220: Errors finding lcm shared library

manuelli, 2017-12-15:
These errors occur when buidling `ContactParticleFilter` or `spartan_lcmtypes`. The error is

```
/home/manuelli/spartan/build/install/bin/lcm-gen: error while loading shared libraries: liblcm.so: cannot open shared object file: No such file or directory
cpf_lcmtypes/lcmtypes/CMakeFiles/lcmtypes_cpf-lcm.sources.dir/build.make:80: recipe for target 'cpf_lcmtypes/lcmtypes/cpf_lcmtypes_actual_and_estimated_contact_locations_t.h' failed
make[6]: *** [cpf_lcmtypes/lcmtypes/cpf_lcmtypes_actual_and_estimated_contact_locations_t.h] Error 127
CMakeFiles/Makefile2:108: recipe for target 'cpf_lcmtypes/lcmtypes/CMakeFiles/lcmtypes_cpf-lcm.sources.dir/all' failed
make[5]: *** [cpf_lcmtypes/lcmtypes/CMakeFiles/lcmtypes_cpf-lcm.sources.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[4]: *** [all] Error 2
CMakeFiles/ContactParticleFilter.dir/build.make:112: recipe for target 'src/ContactParticleFilter-stamp/ContactParticleFilter-build' failed

```

somehow we are having problems with finding the lcm provided by `drake`. The other option is to simply system install lcm and use that.

peteflorence, 2017-12-19:
+1 for running into this issue for a fresh install on my own computer

manuelli, 2017-12-19:
We should investigate why this isn't being caught by the build machine. That's somehow the more serious issue.

gizatt, 2017-12-19:
Is this related to the discussion on
https://github.com/RobotLocomotion/spartan/pull/211? CI caught this error
and it was fixed with an additional required step in the build process.

peteflorence, 2017-12-19:
I'm doing some experimenting, will report back in a second

peteflorence, 2017-12-19:
Okay I was able to build with `ContactParticleFilter` and `spartan_lcmtypes`

It's possible that previously I didn't follow this order in build process:

```
use_ros && cmake .. && use_spartan && make -j8
```

I can now make everything, if I use that order.

gizatt, 2017-12-19:
I suspect

```use_ros && use_spartan && cmake .. && make -j8```

ought to work too. But yeah, `use_spartan` must happen after `cmake` (so
that the script it calls is generated) but before `make`, which must be
able to find lcm-gen that is built and installed by previous steps in
`make.`

manuelli, 2017-12-19:
Ok I may have messed up the `use_spartan` stuff, but I thought I did it correctly. I will investigate as well.

manuelli, 2018-03-15:
Seems like this was due to incorrect build procedure. Closing

closed 2018-03-15 by manuelli