RobotLocomotion/spartan #224: RemoteTreeViewer build fails when updating to drake/master

manuelli, 2017-12-28:
Fails with the following error 

```
[[ 18%] Completed 'drake'
[ 50%] Built target drake
[ 56%] Performing configure step for 'RemoteTreeViewer'
loading initial cache file /home/manuelli/spartan/build/tmp/RemoteTreeViewer-cache-Release.cmake
-- Boost version: 1.58.0
-- Configuring done
CMake Error at CMakeLists.txt:7 (add_library):
  Target "RemoteTreeViewerWrapper" links to target "Boost::boost" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:7 (add_library):
  Target "RemoteTreeViewerWrapper" links to target "Boost::boost" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:15 (add_executable):
  Target "testRemoteTreeViewerWrapper" links to target "Boost::boost" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:15 (add_executable):
  Target "testRemoteTreeViewerWrapper" links to target "Boost::boost" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

googling around makes me think this is due to not having a new enough CMake version.

gizatt, 2017-12-29:
Agreed that versioning is suspect, but we're on CMake version 3.5.1, Boost version 1.58 -- I believe those are pretty commonly paired with each other (e.g. they're both the default off of apt-get, looks like Boost 1.58 was released 2015 and CMake 3.5.1 was 2016, so CMake is sufficiently new that it should know about Boost 1.58.

Almost certainly one of [these](https://github.com/RobotLocomotion/drake/pull/7650/files) changes is the culprit, but I haven't been able to wrap my head around the cause yet, and the low-hanging fruit of `FindPackage(Boost)` and minor variants thereof (that you probably already tried) don't help.

manuelli, 2017-12-29:
Yeah good find on those drake changes. It may be worthing pinging someone at TRI if this is something that could be a quick fix for them.

closed 2018-01-18 by manuelli