RobotLocomotion/spartan #226: docker_build.py failing in drake_install_prereqs

peteflorence, 2018-01-05:
```
Step 18/29 : COPY ./drake/setup/ubuntu/16.04/install_prereqs.sh /tmp/drake_install_prereqs.sh
 ---> Using cache
 ---> 7e8fa9a533a3
Step 19/29 : RUN yes "Y" | /tmp/drake_install_prereqs.sh
 ---> Running in a2e9caf216d5

....

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgl1-mesa-dev : Depends: mesa-common-dev (= 17.0.7-0ubuntu0.16.04.2) but 17.2.4-0ubuntu1~16.04.2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
The command '/bin/sh -c yes "Y" | /tmp/drake_install_prereqs.sh' returned a non-zero code: 100
```

jwnimmer-tri, 2018-01-05:
As of RobotLocomotion/drake#7696, Drake setup no longer tries to explicitly install `mesa-common-dev`.  Hopefully using that fix would resolve this?

peteflorence, 2018-01-05:
Thanks @jwnimmer-tri, did fix the docker build (testing rest now).

Is there something that could have helped avoid, for next time? (not sure what changed)

jwnimmer-tri, 2018-01-05:
I'm not familiar the backstory on that PR, but my impression is that it was a latent hazard in Drake's install script, that came to light when Ubuntu rolled out some updates, so no great ideas for avoiding.

peteflorence, 2018-01-05:
@jwnimmer-tri , do you know if `libbot` has been removed from Drake recently?  

Pointing to Drake master now is causing this error downstream in build: 
```
loading initial cache file /home/peteflo/spartan/build/director/tmp/director-cache-Release.cmake
-- Eigen found (include: /home/peteflo/spartan/build/install/include/eigen3)
-- Eigen found (include: /home/peteflo/spartan/build/install/include/eigen3)
-- Checking for module 'lcm'
--   No package 'lcm' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find LibBot (missing: LIBBOT_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindLibBot.cmake:36 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/app/CMakeLists.txt:213 (find_package)
```

jwnimmer-tri, 2018-01-05:
Yes; see RobotLocomotion/drake#7698; Jamie says Spartan is going to refer to the CMake build of libbot?  In the meantime, if you pin to Drake af0d2ff9bbb8ece23aa59229bfa15fe9041a5063 you will have the setup fix but still keep libbot.

peteflorence, 2018-01-05:
Thanks Jeremy, I've reverted to that commit.

peteflorence, 2018-01-05:
Addressed by https://github.com/RobotLocomotion/spartan/pull/229

closed 2018-01-05 by peteflorence