RobotLocomotion/spartan #230: docker build instructions fail due to no internet access during `docker build`

dmsj, 2018-01-09:
I *think* this occurs due to our somewhat complicated network setup - but when running:
```bash
./setup/docker/docker_build.py
```
The build fails on `RUN apt-get update` with:
```bash
Err:1 http://archive.ubuntu.com/ubuntu xenial InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:2 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1604/x86_64  InRelease
  Temporary failure resolving 'developer.download.nvidia.com'
Err:3 http://security.ubuntu.com/ubuntu xenial-security InRelease
  Temporary failure resolving 'security.ubuntu.com'
Err:4 http://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1604/x86_64  InRelease
  Temporary failure resolving 'developer.download.nvidia.com'
```

This is fixed by editing the build command in `setup/docker/docker_build.py` to add a dns definition:
```python
cmd = "docker build --add-host dns:8.8.8.8 \
			--build-arg USER_NAME=%(user_name)s \
			--build-arg USER_PASSWORD=%(password)s \
			--build-arg USER_ID=%(user_id)s \
			--build-arg USER_GID=%(group_id)s" \
			%{'user_name': user_name, 'password': args.password, 'user_id': args.user_id, 'group_id': args.group_id}
	cmd += " -t %s -f setup/docker/spartan.dockerfile ." % args.image
```
I'm not sure if this will work in general. However it fixed my problem.
@peteflorence

peteflorence, 2018-01-09:
Hey @dmsj, interesting, I don't think anybody has had this problem before, but glad you fixed it.  We can try it on one of our (non-complicated-network) machines and see if it works in general

manuelli, 2018-01-18:
You definitely need internet access during docker build. We pull down lots of packages from `apt-get` and github etc.

closed 2018-01-18 by manuelli