RobotLocomotion/spartan #234: pydrake ik interface broken for kuka software

peteflorence, 2018-01-19:
No worky, can't do teleop, or any other ik calls.  Error relates to pydrake's rbtree.  Potentially related to / fixed by https://github.com/RobotLocomotion/drake/pull/7789

peteflorence, 2018-01-19:
Potentially just need to change semantics on importing rbtree. See @EricCousineau-TRI 's comment in this PR https://github.com/RobotLocomotion/drake/pull/7514/files
```
                 "`import pydrake; pydrake.rbtree` will soon be deprecated." +
                 "\n  Please use `import pydrake.rbtree` or `from pydrake " +
                 "import rbtree` instead.\n")
```

EricCousineau-TRI, 2018-01-19:
Uh oh... What's broken in the IK? `from pydrake import rbtree` should work, it should just spew this warning message (which we can suppress if it's polluting other useful output), but it should still give you the functionality you want to move forward.

Can you point to the error in your code?

EricCousineau-TRI, 2018-01-19:
One note: I ran into https://github.com/RobotLocomotion/drake/issues/7809 - is this related by any chance?

YunzhuLi, 2018-01-20:
Tested with the newest pr #233. IK is still broken. It seems the path referencing the urdf is wrong.
```
Could not open file [/home/yunzhu/spartan/models/iiwa/director/../../../drake/drake/manipulation/models/wsg_50_description/urdf/wsg_50_mesh_collision.urdf] for parsing.
Traceback (most recent call last):
  File "/home/yunzhu/spartan/src/catkin_projects/robot_control/src/robot_control/ikserviceapp.py", line 30, in <module>
    robotSystem = factory.construct(view=fields.view, options=options)
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/componentgraph.py", line 148, in construct
    self.initComponent(name, defaultFields)
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/componentgraph.py", line 164, in initComponent
    newFields = initFunction(inputFields)
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/robotsystem.py", line 171, in initPlanning
    handModels.append(handFactory.getLoader(side))
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 394, in getLoader
    loader = HandLoader(handType, self.robotModel)
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 445, in __init__
    self.loadHandModel()
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 497, in loadHandModel
    handModel = RobotModelItem(handModel)
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/director/roboturdf.py", line 43, in __init__
    modelName = os.path.basename(model.filename())
AttributeError: 'NoneType' object has no attribute 'filename'
```
A simple modification of ```$SPARTAN_SOURCE_DIR/models/iiwa/director/director_config.json``` would solve this issue - changing ```handUrdf``` from
```
"handUrdf" : "../../../drake/drake/manipulation/models/wsg_50_description/urdf/wsg_50_mesh_collision.urdf"
```
to
```
"handUrdf" : "../../../drake/manipulation/models/wsg_50_description/urdf/wsg_50_mesh_collision.urdf"
```

manuelli, 2018-01-20:
I had fixed this locally but must have forgotten to commit it. Sorry. Go
ahead and PR and merge a fix

EricCousineau-TRI, 2018-01-20:
Yes, I think these paths are related to https://github.com/RobotLocomotion/drake/pull/7774 (the issue I posted above was before Jeremy had mentioned this issue to me, and I, uh, hadn't searched well enough).

peteflorence, 2018-01-20:
OK YunZhu has confirmed that IK is OK on master so I'll close this now

closed 2018-01-20 by peteflorence