RobotLocomotion/spartan #236: ROS Trajectory Server is broken

YunzhuLi, 2018-01-20:
Just tested with pr #233, ROS Trajectory Server is broken. Following is the error message from command ```roslaunch robot_control robot_control.launch```:
```
Traceback (most recent call last):
  File "/home/yunzhu/spartan/build/install/lib/robot_control/trajectory_server.py", line 13, in <module>
    from drake import lcmt_iiwa_command, lcmt_iiwa_status ,lcmt_robot_state
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/drake/__init__.py", line 28, in <module>
    from .lcmt_manipulator_plan_move_end_effector import lcmt_manipulator_plan_move_end_effector
  File "/home/yunzhu/spartan/build/install/lib/python2.7/site-packages/drake/lcmt_manipulator_plan_move_end_effector.py", line 12, in <module>
    import bot_core.position_3d_t
ImportError: No module named position_3d_t
```

peteflorence, 2018-01-20:
Hi YunZhu, I am not sure if will help, but have you tried rerunning
docker_build.py after you checked out this commit, then giving another make
-j in spartan/build?

This is another great reason for us to have deeper testing automated by
Jenkins as @gizatt was suggesting yesterday (i.e. move a robot in sim test
in Jenkins)

YunzhuLi, 2018-01-20:
Thanks Pete! Actually I'm starting from a fresh new clone, but still get the above error.

manuelli, 2018-01-20:
Libbot is not built as part of Drake anymore. It's built separately.
bot_core is part of libbot. So just make sure it's on the pythonpath.
Ishould have done more testing before merging. But on the bright side this
is a good learning experience for you.

YunzhuLi, 2018-01-21:
An update on this issue
ROS Trajectory Server cannot be started ```kuka_iiwa_procman``` and is popping out the above error.
Yet ```roslaunch robot_control robot_control.launch``` can be directly executed via terminator, which I'm using now for controlling the arm.

peteflorence, 2018-01-21:
Hi YunZhu, the command that executes from `kuka_iiwa_procman` is exactly this:

https://github.com/RobotLocomotion/spartan/blob/master/apps/iiwa/iiwa_hardware.pmd#L106

Which is the same as the command that you are saying works directly from terminator... hm.

Did you make sure to use_ros / use_spartan in the terminal that you started `kuka_iiwa_procman` from.

YunzhuLi, 2018-01-21:
Yeah. That's why I also think it is weird. Look at the top left that trajectory server is running, but at bottom right that it is not working in ```kuka_iiwa_procman```.
[image]

manuelli, 2018-01-21:
Your pythonoath is messed up. Restart your docker container and do a clean
path sourcing.

manuelli, 2018-01-22:
What's the status on this @YunzhuLi. Did you manage to get it working?

manuelli, 2018-01-22:
Ok so I have made some progress on pinpointing the problem. I added a few print statements to `trajectory_server.py`. 

```
python_path = os.getenv("PYTHONPATH")
print "PYTHONPATH = ", python_path

# drake
import bot_core
print "bot_core.__file__ ", bot_core.__file__
```

When launched from the command line we have

```
PYTHONPATH =  /opt/ros/kinetic/lib/python2.7/dist-packages:/home/manuelli/spartan/build/install/lib/python2.7/site-packages:/home/manuelli/spartan/build/install/lib/python2.7/dist-packages:/home/manuelli/spartan/modules:
bot_core.__file__  /home/manuelli/spartan/build/install/lib/python2.7/site-packages/bot_core/__init__.pyc
```

and everything works fine. But when launched from inside procman we have

```
PYTHONPATH =  /home/manuelli/spartan/build/install/lib/python2.7/dist-packages:/home/manuelli/spartan/build/install/lib/python2.7/site-packages:/opt/ros/kinetic/lib/python2.7/dist-packages:/home/manuelli/spartan/build/install/lib/python2.7/site-packages:/home/manuelli/spartan/build/install/lib/python2.7/dist-packages:/home/manuelli/spartan/modules:
bot_core.__file__  /home/manuelli/spartan/build/install/lib/python2.7/dist-packages/bot_core/__init__.pyc
Traceback (most recent call last):
  File "/home/manuelli/spartan/build/install/lib/robot_control/trajectory_server.py", line 18, in <module>
    from bot_core import robot_state_t
ImportError: cannot import name robot_state_t
```

so notice that now it is finding `bot_core` in `dist-packages`. So somehow bot_core is being installed in two places and depending on whether we launch from the terminal or from `procman` we get two different results. This can be traced to the fact that the `PYTHONPATH` isn't the same in both instances.

If we look in the terminal in which procman was launched we get

```
/opt/ros/kinetic/lib/python2.7/dist-packages:/home/manuelli/spartan/build/install/lib/python2.7/site-packages:/home/manuelli/spartan/build/install/lib/python2.7/dist-packages:/home/manuelli/spartan/modules
```

which is different than what shows up in the print statement inside `trajectory_server` when launched from procman.

PR #239 (2018-01-26): update libbot
fixes #236

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/spartan/239)
closed 2018-01-26 by gizatt