RobotLocomotion/spartan #244: GPU usage at 100% when running pybullet sim

manuelli, 2018-02-08:
This is due to inefficiencies in RemoteTreeViewer. Hacky fix for now is to set a speed limit on the remote tree viewer subscriber. In the `director` python terminal run

```
hz = 2
treeViewer.subscriber.setSpeedLimit(hz)
```

You can also do this programmatically in the launch script.

rdeits, 2018-02-08:
What does pybullet sim do? Is it publishing `setgeometry` every tick?

gizatt, 2018-02-08:
Yuuuup

rdeits, 2018-02-08:
Judging from the Python bindings, it looks like the geometry is probably getting loaded at every tick, which is definitely going to use up your GPU. The right thing to do would be to only do setgeometry once and the use settransform to move the mesh around. That should drastically reduce the GPU load even at high framerates.

rdeits, 2018-02-08:
Ah, ok, yeah. Don't do that :-p

manuelli, 2018-03-15:
@gizatt any progress on this?

gizatt, 2018-03-15:
I know the cause and solution -- would be a few hours to get it done,
though. How high priority is it for you? If its causing trouble I'm happy
to fix it

manuelli, 2018-03-15:
Not high priority at all. Just want to make sure it doesn't fall through the cracks.

manuelli, 2019-03-29:
Have switched to drake sim so this is no longer relevant

closed 2019-03-29 by manuelli