RobotLocomotion/spartan #251: incorporate ROS camera calibration package for intrinsics calibration

manuelli, 2018-02-14:
See http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration. We have decided it is easier to do intrinsics calibration with a handheld target. We will still do extrinsics calibration using our existing setup. 

@YunzhuLi @kmuhlrad would you guys mind taking the lead on this. @peteflorence wants to be involved as well, and I am happy to help as needed.

YunzhuLi, 2018-02-14:
Sure. I think I can take a look into it.

kmuhlrad, 2018-02-14:
I can help on this as well. I’m out for the rest of the day but can sync with @YunzhuLi tomorrow.

manuelli, 2018-02-15:
Ok great. See #252 for a branch that has some updates to work with the new target. Also note that there is a README on the existing calibration routine here https://github.com/RobotLocomotion/spartan/blob/lm-camera-calibration/modules/spartan/calibration/README.md

YunzhuLi, 2018-02-18:
I have gone through the whole pipeline of 
1) using [https://github.com/RobotLocomotion/spartan/blob/lm-camera-calibration/modules/spartan/calibration/README.md](https://github.com/RobotLocomotion/spartan/blob/lm-camera-calibration/modules/spartan/calibration/README.md) to calculate extrinsic parameters
2) using [ http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration]( http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) to get the intrinsic parameters, and then
3) using the new fusion server to check elastic fusion

Following is the result (overlay of live depth clouds and reconstructed clouds), which generally looks good:
[image]

but can still be improved. Mainly my question is
1) if we do intrinsic and extrinsic separately (which are supposed to be jointly optimized in CameraCalibration), how reliable is the final result?
2) Meanwhile, the intrinsic parameters from CameraCalibration is far from satisfying when I went through the pipeline (compare to what we had a week ago), maybe I missed something?

It might be better if we can find a time chatting about all these.

peteflorence, 2018-02-18:
Hey YunZhu, glad you went through the calibration.  Is that on Kuka #2?

It may have been better to switch your order of 1 and 2, ie:
1) use ROS to get intrinsics
2) keep intrinsics set, change Camera Calibration to only optimize for extrinsics

And glad you also checked with ElasticFusion server.

Agreed that table doesn't look as good as it could.  But it's pretty OK.

This coming week for me, top priorities for me research-wise will be locking in calibration, tweaking fusion, and collecting good data.  So yes I hope to be able to contribute a lot to improving those. I'll be a little tied up next day or so preparing the homework for 6.832 though.

Yes let's chat soon!

YunzhuLi, 2018-02-18:
Hi Pete,

Yes. That's kuka 2 with the old Xtion PRO LIVE.

Sure, let's chat soon! I will be around as long as you are available.

kmuhlrad, 2018-02-19:
Was the intrinsics calibration done within spartan, or separately? I've been trying to get the camera_calibration package within spartan with no luck.

YunzhuLi, 2018-02-19:
within spartan. You might need to install a ros package called [image_pipeline](http://wiki.ros.org/image_pipeline).

manuelli, 2018-02-20:
Yes Pete is right in that we should pass the intrinsics computed using the ROS package to the CameraCalibration routine. Just to clarify the CameraCalibration package was not jointly optimizing intrinsics and extrinsics. It would compute intrinsics first, and then with those intrinsics fixed, compute the extrinsics.

@YunzhuLi do you have a branch that has the image_pipeline package integrated? Is there a README somwhere. If you have a required package go ahead and add it to the prereqs for the docker build.

kmuhlrad, 2018-02-20:
I have a [branch](https://github.com/kmuhlrad/spartan/tree/camera_calibration) where I try to pull in the ros-kinetic-image-pipeline package into spartan through the install scripts, although it's not currently working. However it might be a helpful place to start if you aren't already adding it to the docker build.

kmuhlrad, 2018-02-20:
This branch has been merged into master, so @YunzhuLi, any scripts you were running within spartan relying on camera_calibration should work with an update and clean build.

peteflorence, 2018-02-25:
Calibration is under control for now, thanks everybody

Closing this issue

We do need to work a bit on documentation but one step at a time

closed 2018-02-25 by peteflorence