RobotLocomotion/spartan #266: Director doesn't build with drake/master

manuelli, 2018-03-14:
Attempting to build `spartan` with `drake/master`. The `drake` sha is `d4b3bde6de2b89f21ad2fe5df998c069dbd9407c`. 

Drake builds fine but then director has the build error. Looks like the vtk version that `drake` is providing is no longer compatible with `director`. Has the vtk version changed recently?

@patmarion @fbudin69500 

```
/home/manuelli/spartan/build/director/src/pcl/surface/src/vtk_smoothing/vtk_utils.cpp: In static member function 'static int pcl::VTKUtils::vtk2mesh(const vtkSmartPointer<vtkPolyData>&, pcl::PolygonMesh&)':
/home/manuelli/spartan/build/director/src/pcl/surface/src/vtk_smoothing/vtk_utils.cpp:124:20: error: 'class vtkUnsignedCharArray' has no member named 'GetTupleValue'
       poly_colors->GetTupleValue (i, &point_color[0]);
                    ^
/home/manuelli/spartan/build/director/src/pcl/surface/src/vtk_smoothing/vtk_utils.cpp: In static member function 'static int pcl::VTKUtils::mesh2vtk(const pcl::PolygonMesh&, vtkSmartPointer<vtkPolyData>&)':
/home/manuelli/spartan/build/director/src/pcl/surface/src/vtk_smoothing/vtk_utils.cpp:240:15: error: 'class vtkUnsignedCharArray' has no member named 'InsertNextTupleValue'
       colors->InsertNextTupleValue (color);
               ^
/home/manuelli/spartan/build/director/src/pcl/surface/src/vtk_smoothing/vtk_utils.cpp:257:16: error: 'class vtkFloatArray' has no member named 'InsertNextTupleValue'
       normals->InsertNextTupleValue (normal);

```

sammy-tri, 2018-03-14:
Yes, VTK in drake update recently (I can't find the PR quickly).  I've hit the same problem.  VTK was then rolled back to an older version in #8325, but it wasn't the same older version and these deprecated functions are still missing (and director thus still fails to compile).   cc @jamiesnape

jamiesnape, 2018-03-14:
Yes, `GetTupleValue` needs to be changed to `GetTypedTuple` and `InsertNextTupleValue` needs to be changed to `InsertNextTypedTuple`.

jamiesnape, 2018-03-14:
https://github.com/PointCloudLibrary/pcl/blob/fc88d45bcc4f49904bcff1b4359f1310ab516292/surface/src/vtk_smoothing/vtk_utils.cpp#L52-L57

manuelli, 2018-03-15:
@patmarion is it an easy fix to `director` to incorporate https://github.com/PointCloudLibrary/pcl/blob/fc88d45bcc4f49904bcff1b4359f1310ab516292/surface/src/vtk_smoothing/vtk_utils.cpp#L52-L57?

patmarion, 2018-03-16:
Director is currently building with pcl 1.8.0:

https://github.com/RobotLocomotion/director/blob/master/distro/superbuild/cmake/externals.cmake#L472

You could try changing that to 1.8.1.  If it builds then I think we're good.

sammy-tri, 2018-03-16:
Tried changing director to build with the current drake head.  PCL built, but then director blew up looking for LCM, presumably because drake doesn't provide a `.pc` file for it:

-- Checking for module 'lcm'
--   No package 'lcm' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find LCM (missing: LCM_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindLCM.cmake:29 (find_package_handle_standard_args)
  cmake/modules/FindLCM.cmake:38 (pkg_config_find_module)
  src/app/CMakeLists.txt:183 (find_package)

Will investigate a bit more.

sammy-tri, 2018-03-16:
Deleting `cmake/modules/FindLCM.cmake` got me further along the build, but PointCloudLibraryPlugin has the same problem with `SetTupleValue`.

jamiesnape, 2018-03-16:
I can PR a fix up to `PointCloudLibraryPlugin` this afternoon.

jamiesnape, 2018-03-19:
Try updating the director build to use https://github.com/patmarion/PointCloudLibraryPlugin/commit/960f5ff1d3fe31e1a457d9566408ce3cde90df62.

sammy-tri, 2018-03-19:
Thanks Jamie!  Almost worked.  Opened https://github.com/patmarion/PointCloudLibraryPlugin/pull/5 which gets PointCloudLibraryPlugin compiling with the vtk installed by https://github.com/RobotLocomotion/drake/commit/c55d4fdf1ddfbdbf67d26e9fcc2290e5ac81d502

sammy-tri, 2018-03-19:
I have a director tree compiling in spartan now: https://github.com/sammy-tri/director/tree/drake_fixups_20180319

Highlights are roughly what's been mentioned above...   bump PCL version, build `PointCloudPluginLibrary`, delete `FindLCM.cmake`.  It hasn't been exhaustively tested beyond loading my fork of the iiwa app.

Compilation options: `-DWITH_IIWA_DRIVER_TRI=OFF  -DWITH_IIWA_DRIVER_RLG=OFF -DWITH_PERCEPTION=ON -DWITH_SNOPT=ON -DWITH_ROS=OFF`

I can open a PR to director if we think this is a reasonable solution.  I don't know if I'm going to break any other workflows by changing how LCM is found.

jamiesnape, 2018-03-20:
PR the two SHA bumps and meanwhile I will look into the LCM issues.

sammy-tri, 2018-03-20:
PR opened.  https://github.com/RobotLocomotion/director/pull/596

sammy-tri, 2018-03-20:
director PR now merged, only the LCM issue should remain

jamiesnape, 2018-03-22:
Patch related to LCM merged, so I think we should be good now.

closed 2018-06-22 by manuelli