RobotLocomotion/spartan #300: Drake iiwa simulation

manuelli, 2018-06-19:
- [ ] Ensure it is wired up in the same way that the robot hardware is in terms of messages etc.
- [ ] Simulate depth (more important) and RGB (less important) messages.
- [ ] Easy interface for adding/removing objects to the scene. We should be able to easily have a table, a bin, a few objects etc.

This simulator should live in spartan, probably as a catkin project. Should target drake as a library.

manuelli, 2018-07-12:
@pangtao22 has made great progress on getting our drake-iiwa-sim back up and running. Thanks! @gizatt I think if we could get the depth images to be rendered through the drake sim and then published out as ROS messages that would be great. Can you take the lead on that?

gizatt, 2018-07-12:
Yeah glad to.