RobotLocomotion/spartan #308: Flaky test simuluation_test::IiwaSimulationTest::test_move_arm

gizatt, 2018-06-26:
Passed: http://spartan-jenkins.csail.mit.edu:8080/job/spartan-master/412/
Failed: http://spartan-jenkins.csail.mit.edu:8080/job/spartan-master/411/

On the same underlying code. Looks like maybe roscore failed to start up?

For now the patch can be to manually re-trigger the appropriate build so we can collect a little more data on how frequently this happens. If it becomes a serious problem, that test can be disabled.

manuelli, 2018-06-26:
Yeah I don't think I did a great job setting up the test. There is a 3 second sleep somewhere in that test. Maybe we need to make that longer. I also have a part of the code where I kill all processes with `procman` in the name. Neither of these are elegant solutions, but they were what I came up with at the time. Happy to modify as needed.

gizatt, 2018-08-30:
This seems to have gotten better with the smarter procman-kill-everything fixes in the test scripts.

closed 2018-08-30 by gizatt