RobotLocomotion/spartan #311: tsdf-fusion not in build system

gizatt, 2018-07-10:
Requires working out CUDA in CMake -- or is there another reason for this?

manuelli, 2018-07-11:
This is just because it got brought in quickly by @peteflorence and myself. It uses a makefile which runs just fine from inside the docker container. Happy to have this be part of the main build system.

peteflorence, 2018-07-11:
Hey guys yes is just a ‘make’ from inside that folder. Unless there is
desire to link against it I think easiest to just leave it there, but
whatever works

gizatt, 2018-07-11:
That utility (and the fusion server y'all wrote) was super useful for getting pretty scans of some scenes I want to work on -- I'll try to drag this into the build system some time to cement it, and hopefully drop some more documentation / a few fixes in when I do that.

manuelli, 2018-11-05:
Andy Zeng gave us a new python only version of this functionality. I think switching to that is the best functionality.