RobotLocomotion/spartan #312: Add support for visualizing ROS pointclouds in director

manuelli, 2018-07-13:


patmarion, 2018-07-13:
director/master includes a top level `catkin_ws` folder.  first build director as usual, then build the catkin_ws.  the director superbuild does this automatically when you enable `USE_ROS` see:

https://github.com/RobotLocomotion/director/blob/master/distro/superbuild/cmake/externals.cmake#L13

and:

https://github.com/RobotLocomotion/director/blob/master/distro/superbuild/cmake/externals.cmake#L642

note in the above code, after building the catkin_ws, it "installs" it by copying the libddROSPlugin.so file.  To learn more about the details, look at the source code of the c++ code that was compiled to make that plugin.  To run the plugin, here is a simple python snippet that can be added to any existing director code:

https://github.com/RobotLocomotion/director/blob/master/catkin_ws/src/director_ros/src/pointcloud_subscriber.py

The plugin starts a ROS node and spins it on a separate thread.  The thread receives pointcloud messages and converts them to vtkPolyData objects, then it notifies Python that a new object is available.  The python main thread grabs the vtkPolyData from the c++ thread and takes ownership.

If this is useful, I also have image subscriber code I can share.

patmarion, 2018-07-13:
Oops, in the above post, please ignore the `pointcloud_subscriber.py` file I linked to.  That file is out of date and should be deleted.  This is the more comprehensive example file:

https://github.com/RobotLocomotion/director/blob/master/src/python/tests/testRosPlugin.py