RobotLocomotion/spartan #339: build error looking for wsg 50 msgs

peteflorence, 2018-10-04:
@gizatt @manuelli 

Hey guys, I believe temporarily fixed this by nuking `drake-schunk-driver`, which seems to be vestigial?  Potentially may be worth fully removing from git?  There are a handful of tricky steps including removing from .gitmodules, see below.  I imagine this doesn't happen on new builds, since Jenkins passes

https://github.com/RobotLocomotion/spartan/blob/d868924c6b348566c1fc130d5d84a519fc137b57/.gitmodules#L19

gizatt, 2018-10-05:
Thought I had fixed this, but yeah, looks like it but needs to be cleaned from `.gitmodules`. In the mean time, manually removing the `drake-schunk-driver` folder fixes this... and it shouldn't occur on new, clean clones of the repo, agreed.

gizatt, 2018-10-10:
Resolved, I think, by PR#340. LMK if not?

closed 2018-10-10 by gizatt