RobotLocomotion/spartan #341: Submodule update fails on submodules: iiwa_driver and handical

iskanderuse, 2018-10-05:
During quickstart in section 3 the command `git submodule update --init` fails.
The submodules **iiwa_driver** and **handicap** doesn't exists in repositories.

manuelli, 2018-10-05:
As stated in the README this repository contains private submodules (which are exactly `iiwa_driver` and `handical`) and is not supported for external use (i.e. outside our research group).

closed 2018-10-05 by manuelli

peteflorence, 2018-10-05:
@iskanderuse, :) sorry this repo isn't directly usable for you out of the box.  There's just a couple submodules that contain proprietary code (not written by us).

The code is maybe still useful for you though, as reference for how our system is put together.  It's all BSD-3 license if you want to grab any part of it and use it.  Would require significant work to integrate though.