RobotLocomotion/spartan #377: no such package '@snopt//'

kingsman142, 2019-05-28:
I'm currently following this guide: https://github.com/RobotLocomotion/spartan/blob/master/setup/docker/README.md and I made it to the last line where i have to `make -j` spartan. However, I receive the following two errors and my build fails:

[ 23%] Built target pybind11
INFO: Invocation ID: f20c81ea-ba25-430b-8859-f69b9d94b41a
ERROR: /home/james/spartan/drake/solvers/BUILD.bazel:735:1: no such package '@snopt//': Traceback (most recent call last):
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 150
		_setup_git(repo_ctx)
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 60, in _setup_git
		_git_clone(repo_ctx, remote = repo_ctx.attr.r..., ...)
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 47, in _git_clone
		_run_git(repo_ctx, "clone", remote, ".")
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 27, in _run_git
		_execute(repo_ctx, "git", (["git", "--git-dir...)))
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 16, in _execute
		fail("Repository rule @{} error {} du...))))
Repository rule @snopt error 128 during git operation: "Cloning into '.'...\nERROR: Repository not found.\nfatal: Could not read from remote repository.\n\nPlease make sure you have the correct access rights\nand the repository exists.\n" and referenced by '//solvers:snopt_solver'
ERROR: Analysis of target '//:install' failed; build aborted: no such package '@snopt//': Traceback (most recent call last):
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 150
		_setup_git(repo_ctx)
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 60, in _setup_git
		_git_clone(repo_ctx, remote = repo_ctx.attr.r..., ...)
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 47, in _git_clone
		_run_git(repo_ctx, "clone", remote, ".")
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 27, in _run_git
		_execute(repo_ctx, "git", (["git", "--git-dir...)))
	File "/home/james/spartan/build/drake/_bazel_/6a242f2d6d1177c198b7f921704eb9bb/external/drake/tools/workspace/snopt/repository.bzl", line 16, in _execute
		fail("Repository rule @{} error {} du...))))
Repository rule @snopt error 128 during git operation: "Cloning into '.'...\nERROR: Repository not found.\nfatal: Could not read from remote repository.\n\nPlease make sure you have the correct access rights\nand the repository exists.\n"
INFO: Elapsed time: 5.725s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (209 packages loaded, 15660 targets configured)
    Fetching @bullet; fetching
    Fetching @json; fetching
    Fetching @eigen; fetching
    Fetching @vtk; fetching
    Fetching @fcl; fetching
    Fetching @drake_visualizer; fetching
    Fetching @org_apache_xmlgraphics_commons; fetching
    Fetching @remotejdk_linux; fetching ... (11 fetches)
CMakeFiles/drake_cxx_python.dir/build.make:110: recipe for target 'drake_cxx_python-prefix/src/drake_cxx_python-stamp/drake_cxx_python-build' failed
make[5]: *** [drake_cxx_python-prefix/src/drake_cxx_python-stamp/drake_cxx_python-build] Error 1
CMakeFiles/Makefile2:387: recipe for target 'CMakeFiles/drake_cxx_python.dir/all' failed
make[4]: *** [CMakeFiles/drake_cxx_python.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/drake.dir/build.make:111: recipe for target 'src/drake-stamp/drake-build' failed
make[2]: *** [src/drake-stamp/drake-build] Error 2
CMakeFiles/Makefile2:179: recipe for target 'CMakeFiles/drake.dir/all' failed
make[1]: *** [CMakeFiles/drake.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

manuelli, 2019-05-28:
@kingsman142 Unfortunately `spartan` is really a tool meant for internal lab use (as stated in the main README), not for use by the broader community. I think your issue arises from insufficient permissions

closed 2019-05-28 by manuelli

Amulya21, 2019-07-17:
Hii,
Even i am facing the same issue.what to do? How to resolve this issue?